Nefry-Community/arduino-esp32 #84: 1.1.3のライブラリだとスケッチ書き込みが出来ない

n0bisuke, 2018-01-05:
### Hardware:
Board:							Nefry BT R2
Core Installation/update date:			2018/1/5
IDE name:							Arduino IDE 1.8.3
Flash Frequency:					?
Upload Speed:						921600

### Description:

1.1.3のライブラリでスケッチを書き込もうとしたらエラーになりました。

* macOS High Sierra

### Sketch:

空のプログラム

```cpp
void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Archiving built core (caching) in: /var/folders/xz/gdm2v3kd2d9cbcv2ncnvn7rc0000gn/T/arduino_cache_885371/core/core_Nefry(ESP32)_esp32_nefrybtr2_UploadSpeed_921600,DebugLevel_none_7b901791008b538cc40c8e9fd25cfe03.a
最大1703936バイトのフラッシュメモリのうち、スケッチが566401バイト（33%）を使っています。
最大294912バイトのRAMのうち、グローバル変数が44404バイト（15%）を使っていて、ローカル変数で250508バイト使うことができます。
スケッチの書き込み中にエラーが発生しました
```

n0bisuke, 2018-01-05:
Arduino IDE 1.8.5にしたら別のメッセージが

```
Archiving built core (caching) in: /var/folders/xz/gdm2v3kd2d9cbcv2ncnvn7rc0000gn/T/arduino_cache_741357/core/core_Nefry(ESP32)_esp32_nefrybtr2_UploadSpeed_921600,DebugLevel_none_7b901791008b538cc40c8e9fd25cfe03.a
Index error: could not find referenced tool name=esptool version=4dab24e packager=Nefry(ESP32)
最大1703936バイトのフラッシュメモリのうち、スケッチが566401バイト（33%）を使っています。
最大294912バイトのRAMのうち、グローバル変数が44404バイト（15%）を使っていて、ローカル変数で250508バイト使うことができます。
スケッチの書き込み中にエラーが発生しました
```

wamisnet, 2018-02-21:
#85  同一

closed 2018-02-21 by wamisnet