Nefry-Community/arduino-esp32 #85: 書き込み途中で失敗する

n0bisuke, 2018-01-05:
```
Arduino：1.8.5 (Mac OS X), ボード："Nefry BT R2, 921600, None"

Archiving built core (caching) in: /var/folders/xz/gdm2v3kd2d9cbcv2ncnvn7rc0000gn/T/arduino_cache_741357/core/core_Nefry(ESP32)_esp32_nefrybtr2_UploadSpeed_921600,DebugLevel_none_cd0c8997c752abf476cc6e1151acf722.a
最大1703936バイトのフラッシュメモリのうち、スケッチが548671バイト（32%）を使っています。
最大294912バイトのRAMのうち、グローバル変数が44808バイト（15%）を使っていて、ローカル変数で250104バイト使うことができます。
esptool.py v2.1-beta1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4348.2 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 11120 bytes to 7467...

Writing at 0x00001000... (100 %)
Wrote 11120 bytes (7467 compressed) at 0x00001000 in 0.1 seconds (effective 928.1 kbit/s)...
Hash of data verified.
Compressed 549824 bytes to 346038...

Writing at 0x00010000... (4 %)
Writing at 0x00014000... (9 %)
Writing at 0x00018000... (13 %)
Writing at 0x0001c000... (18 %)
Writing at 0x00020000... (22 %)
Writing at 0x00024000... (27 %)
Writing at 0x00028000... (31 %)
Writing at 0x0002c000... (36 %)
A fatal error occurred: Failed to write compressed data to flash after seq 7 (result was C100)
A fatal error occurred: Failed to write compressed data to flash after seq 7 (result was C100)

「ファイル」メニューの「環境設定」から
「より詳細な情報を表示する：コンパイル」を有効にすると
より詳しい情報が表示されます。
```

n0bisuke, 2018-01-05:
[image]

n0bisuke, 2018-01-05:
何回か書き込み繰り返すとたまにうまくいきます。

wamisnet, 2018-01-05:
謎だなぁ

wamisnet, 2018-02-21:
コンデンサを外すことを紹介する感じで進める

closed 2018-04-20 by wamisnet