Nefry-Community/arduino-esp32 #91: fastsensingのライブラリを利用すると途中で落ちて再起動がかかる

n0bisuke, 2018-01-18:
タイトルの通り、起動中に数回リクエストしたあとに途中で落ちます。

* Nefry BT R2
* ライブラリ 1.1.4
* Arduino IDE 1.8.5
* macos Hight Sierra

コンパイルや書き込みではなく実行してからのシリアルモニタでのログです。

```
assertion "conn->state == NETCONN_CONNECT" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/api_msg.c", line 1230, function: lwip_netconn_do_connected
abort() was called at PC 0x400de397 on core 1

Backtrace: 0x40087ca4:0x3ffd7150 0x40087da3:0x3ffd7170 0x400de397:0x3ffd7190 0x4011dcd6:0x3ffd71c0 0x40112881:0x3ffd71e0 0x40113311:0x3ffd7200 0x401175d2:0x3ffd7220 0x4011cb8a:0x3ffd7240 0x4010c79d:0x3ffd7260

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c
```

n0bisuke, 2018-01-18:
書き込みプログラムはこちら

```
#include <NefryFastSensing.h>
//インスタンスを作成します
NefryFastSensing fastSensing;
float floatData = 0;
float intData = 0;

void setup() {
  //FastSensingで必要な情報を入力し、初期化をします
  //https://console.fastsensing.com/devicesからデバイスとチャンネル
  //begin(デバイス、チャンネル1、チャンネル2、チャンネル3)
  fastSensing.begin("device", "channel1", "channel2", "channel3");
}

void loop() {
  //FastSensingに送るデータを作成します
  //setValue(チャンネル、データ)
  fastSensing.setValue(0, floatData);
  fastSensing.setValue(1, intData);

  //setValueで入れた値をFastSensingに送信します
  //push()
  fastSensing.push();
  
  floatData += 0.1;
  intData++;
  delay(10000);

}
```

wamisnet, 2018-01-18:
https://qiita.com/matobaa/items/cc92fb105bdb5cdbe6f3
この記事を参考にスタックトレースの解析をしてもいいですか

wamisnet, 2018-01-28:
@n0bisuke こちらCloseしても大丈夫ですか？

closed 2018-01-30 by wamisnet