Nefry-Community/arduino-esp32 #96: NeryBTr2no

wamisnet, 2018-02-07:


wamisnet, 2018-02-21:
周波数がちがうのかなぁ

wamisnet, 2018-03-11:
```
#include <Nefry.h>
#include <NefryDisplay.h>
//フルカラーLED　ランダムに色が変わります。
void setup() {
  NefryDisplay.begin();
  Nefry.setProgramName("FullColorLED"); //プログラム名を管理することができます。
}
void loop() {
  NefryDisplay.print("Red");
  Nefry.setLed(255, 0, 0);       //LEDがランダムに点灯します。
  delay(5000);                          //1秒待つ
  NefryDisplay.print("Green");
  Nefry.setLed(0, 255, 0);       //LEDがランダムに点灯します。
  delay(5000);                          //1秒待つ
  NefryDisplay.print("Blue");
  Nefry.setLed(0, 0, 255);       //LEDがランダムに点灯します。
  delay(5000);                          //1秒待つ
  NefryDisplay.print("OFF");
  Nefry.setLed(0, 0, 0);       //LEDがランダムに点灯します。
  delay(5000);                          //1秒待つ
}
```

1ft-seabass, 2018-03-12:
1.2.0でやってみました。以下のとおりです。

* ソースそのまま
  * ディスプレイあり → Red・Green・Blue・OFF 正常稼働
  * ディスプレイなし → Red・Green・Blue・OFF 正常稼働
* ソースからNefryDisplay関連の総コメントアウト
  * ディスプレイあり → Red・Green・Blue・OFF 正常稼働
  * ディスプレイなし → Red・Green・Blue正常稼働　消灯しない

ソースからNefryDisplay関連の総コメントアウトの内容は以下です。

```c
#include <Nefry.h>
// #include <NefryDisplay.h>
//フルカラーLED　ランダムに色が変わります。
void setup() {
  // NefryDisplay.begin();
  Nefry.setProgramName("FullColorLED"); //プログラム名を管理することができます。
}
void loop() {
  // NefryDisplay.print("Red");
  Nefry.setLed(255, 0, 0);       //LEDがランダムに点灯します。
  delay(5000);                          //1秒待つ
  // NefryDisplay.print("Green");
  Nefry.setLed(0, 255, 0);       //LEDがランダムに点灯します。
  delay(5000);                          //1秒待つ
  // NefryDisplay.print("Blue");
  Nefry.setLed(0, 0, 255);       //LEDがランダムに点灯します。
  delay(5000);                          //1秒待つ
  // NefryDisplay.print("OFF");
  Nefry.setLed(0, 0, 0);       //LEDがランダムに点灯します。
  delay(5000);                          //1秒待つ
}
```

1ft-seabass, 2018-03-12:
すいません、ちょっと「ソースからNefryDisplay関連の総コメントアウト」のほう間違ってました。両方、消灯しませんでした。

* ソースそのまま
  * ディスプレイあり → Red・Green・Blue・OFF 正常稼働
  * ディスプレイなし → Red・Green・Blue・OFF 正常稼働
* ソースからNefryDisplay関連の総コメントアウト
  * ディスプレイあり → Red・Green・Blue正常稼働　消灯しない（前コメントが間違い）
  * ディスプレイなし → Red・Green・Blue正常稼働　消灯しない

closed 2018-03-12 by wamisnet