Nefry-Community/arduino-esp32 #102: R3: Groveアナログポートが読み取れない

n0bisuke, 2018-03-16:
（どのピンがだめが表にまとめてもらえると助かります）
Nefry BT r2でA2A3がだめなのかも気になります

n0bisuke, 2018-03-16:
* Grove A0

```
#define LIGHT_SIG A0

void setup() {
  Serial.begin(115200);                //Start the Serial connection
  pinMode(LIGHT_SIG, INPUT);
}

void loop() {
  int sensorValue = analogRead(LIGHT_SIG);
  Serial.println(sensorValue);
  delay(1000);
}
```

コンパイルは成功

Lightセンサーをつなげても値が正常に取れない

[image]

n0bisuke, 2018-03-16:
A2にさしたら通常通り動きましたww

n0bisuke, 2018-03-16:
現状だとA0とA2が逆に配線されてしまっている模様

## A0に指す場合

```
#define LIGHT_SIG A2

void setup() {
  Serial.begin(115200);                //Start the Serial connection
  pinMode(LIGHT_SIG, INPUT);
}

void loop() {
  int sensorValue = analogRead(LIGHT_SIG);
  Serial.println(sensorValue);
  delay(1000);
}
```

## A2に指す場合

```
#define LIGHT_SIG A0

void setup() {
  Serial.begin(115200);                //Start the Serial connection
  pinMode(LIGHT_SIG, INPUT);
}

void loop() {
  int sensorValue = analogRead(LIGHT_SIG);
  Serial.println(sensorValue);
  delay(1000);
}
```

wamisnet, 2018-03-16:
ハードウエア側の問題、修正ずみ

closed 2018-03-16 by wamisnet